omerfaruk-aran/esphome_samsung_hvac_bus #191: [nonNASA] No communication established is AC and ESP started together.

tristone-cz, 2024-10-15:
Hi.

I am playing with the AC about last week. Handled several issues, including missing 12V, never mind.

Today I connected power supply 230V -> 12V and the ESP works.
But in overall the setup behaves weird.

If I turn on power for AC (not start eg. cooling, really connect power) the ESP cannot start communication with the AC.
There are endless attempts to register as controller:
```
INFO Successfully connected to ac-rooms @ 192.168.11.211 in 0.007s
INFO Successful handshake with ac-rooms @ 192.168.11.211 in 0.105s
[10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
...
[10:58:25][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
[10:58:25][W][samsung_ac:175]: NonNASA: invalid crc - got 112 but should be 118: 3200c8fc3f0b343201c8fc457034
[10:58:25][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
```
It does not stop even in 10 minutes. Just from time to time with a message of invalid CRC.

If I just disconnect power from ESP and connect back, no difference.

BUT. If I power on the AC without ESP powered and then in a minute or two power on the ESP - all works. 
Communication works, controls work.

I tried many times, some combinations what to put on and off etc. 
The conclusion is that only sequence which works is `Power AC -> delay -> Power ESP`
Once everything runs fine it also works to power off the AC while keeping ESP up and power it back. But this is quite useless scenario.

Now when the AC is opened it is manageable but once covered there will be no way how to recover from power loss.

Have you seen anything like that? Any idea how this can be caused?
Maybe to add some delay to startup after power on ....

lanwin, 2024-10-16:
Hi @tristone-cz that looks wired. The CRC errors are normal. I also have this in my NASA system. I just leave them to have a better understanding how often it happens.

Did you tried to turn on message logging as described in readme?

tristone-cz, 2024-10-16:
Hi @lanwin I tried now the logging.

Log number **3** is just basic logging, directives `debug_log_` to true.
It is even just upload new FW and ESP restart, no power off of the AC.

Log number **4** is a bit interesting. Slightly different behavior, I saw once already. But not regular.
I turned on the UART debug, deploy FW, restart ESP. No power off.
At 14:50:35 there is an understood message with status of the AC. Since that time the error `Unsupported alt_mode 0` is present which is a sign of somehow working communication. But still many of the those registration messages.
At this point it is possible to control the AC and once first command is successfully sent the registration messages stop. At time 14:51:05.
But this behavior is random. I saw it only twice. In other cases no communication.

Log number **5** is with the same debug ar log 4. Just AC was powered off and then on which turns on also the ESP.
And no communication established.

[logs_ac-rooms_run_3.txt](https://github.com/user-attachments/files/17396149/logs_ac-rooms_run_3.txt)
[logs_ac-rooms_run_4.txt](https://github.com/user-attachments/files/17396148/logs_ac-rooms_run_4.txt)
[logs_ac-rooms_logs_5.txt](https://github.com/user-attachments/files/17396147/logs_ac-rooms_logs_5.txt)

tristone-cz, 2024-10-24:
I managed to handle the delay.
I forked the repo and added delay for first registration. And with the delay it works. Even somehow the registration is not needed, the communication starts sooner then the delay expires.

What I failed is to add a config parameter so I did not create pull request yet. So far no idea what I missed 😢 

Code comparison:
https://github.com/lanwin/esphome_samsung_hvac_bus/compare/main...tristone-cz:esphome_samsung_ac:tristone-cz-patch-1

That code does not take the parameter into account in `protocol_non_nasa.cpp`

The commented out line work with fixed delay of 1 minute.
```
	//const long delay_interval = (1 *60 * 1000UL);
	const long delay_interval = (non_nasa_registration_delay_interval * 1000UL);
```

tristone-cz, 2025-04-03:
> > Hi.
> > I am playing with the AC about last week. Handled several issues, including missing 12V, never mind.
> > Today I connected power supply 230V -> 12V and the ESP works. But in overall the setup behaves weird.
> > If I turn on power for AC (not start eg. cooling, really connect power) the ESP cannot start communication with the AC. There are endless attempts to register as controller:
> > ```
> > INFO Successfully connected to ac-rooms @ 192.168.11.211 in 0.007s
> > INFO Successful handshake with ac-rooms @ 192.168.11.211 in 0.105s
> > [10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:12][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:13][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > ...
> > [10:58:25][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > [10:58:25][W][samsung_ac:175]: NonNASA: invalid crc - got 112 but should be 118: 3200c8fc3f0b343201c8fc457034
> > [10:58:25][W][samsung_ac:091]: write 32d0c8d1d2000000000000001b34
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > It does not stop even in 10 minutes. Just from time to time with a message of invalid CRC.
> > If I just disconnect power from ESP and connect back, no difference.
> > BUT. If I power on the AC without ESP powered and then in a minute or two power on the ESP - all works. Communication works, controls work.
> > I tried many times, some combinations what to put on and off etc. The conclusion is that only sequence which works is `Power AC -> delay -> Power ESP` Once everything runs fine it also works to power off the AC while keeping ESP up and power it back. But this is quite useless scenario.
> > Now when the AC is opened it is manageable but once covered there will be no way how to recover from power loss.
> > Have you seen anything like that? Any idea how this can be caused? Maybe to add some delay to startup after power on ....
> 
> Hi, can you share your setup with me?

HI.

I am using quite a standard setup.
The only difference is to use my own fork which contains the suggested fix as it is not yet merged to main branche.

These are the interesting lines. And in fact the really important is just the source 😃 
```
external_components:
  - source: github://tristone-cz/esphome_samsung_hvac_bus@patch-slow-registration  
    components: [samsung_ac]
    refresh: 0s

samsung_ac:
  non_nasa_keepalive: true
  debug_log_undefined_messages: true
  debug_log_messages: true
```

wifi75, 2025-04-03:
I'm trying it but it doesn't work for me, I have only one external unit and 2 internal ones.

[image]

wifi75, 2025-04-03:
[image]

tristone-cz, 2025-04-03:
Is it trying to connect to the devices? I have the same setup, 1 outdoor, 2 indoor.
It shall send a registration message every two seconds and in about 20-30 seconds by latest get response from the outdoor unit.

I have troubles now to catch the startup logs online. Before the logs get available the initial sequence is already gone and ESP is communicating with AC :-/

Do you see any messages coming once to turn on the debug? Eg. if you do something on remote? You shall see all messages on the bus even if ESP is not registered to the controller.

wifi75, 2025-04-03:
> messaggi sul bus anche se ESP non è registrato sul controller.

if I debug I don't see anything if for example I perform operations from the remote control...

could it be a problem with inverted f1 and f2 cables?
Could it be?

tristone-cz, 2025-04-03:
With these two lines
```
  debug_log_undefined_messages: true
  debug_log_messages: true
```
you shall see all the traffic on the bus. 

Even to remote control shall not be needed, the indoor and outdoor units talks quite a lot themselves.
The log shall not be calm.

As it is simple RS485 bus all messages comes to everyone. If there is silent you are almost sure incorrectly connected.

wifi75, 2025-04-03:
I don't see anything.... I don't understand... what should I check?

tristone-cz, 2025-04-03:
Two things - correct F1/F connectivity and then proper configuration of UART section at ESPhome.
You have to have correct PINs in UART and these are different for different ESP boards

wifi75, 2025-04-03:
> Two things - correct F1/F connectivity and then proper configuration of UART section at ESPhome. You have to have correct PINs in UART and these are different for different ESP boards

please can you post your full configuration?

wifi75, 2025-04-03:
In my config there isn' UART section at ESPhome, can you hellp me please?

tristone-cz, 2025-04-03:
> > Two things - correct F1/F connectivity and then proper configuration of UART section at ESPhome. You have to have correct PINs in UART and these are different for different ESP boards
> 
> please can you post your full configuration?

```
esphome:
  name: ac-rooms
  friendly_name: ac-rooms

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 8MB
#  framework:
#    type: esp-idf

web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR # Remove the "Your component takes to long to respond warning"

api:
  encryption:
    key: "XXX"

ota:
  - platform: esphome
    password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: "192.168.yyy"
  ap:
    ssid: "Ac-Rooms Fallback Hotspot"
    password: "XXX"

captive_portal:
    
uart:
  tx_pin: GPIO06   # Use GPIO26 for ATOM Tail485 
  rx_pin: GPIO05   # Use GPIO32 for ATOM Tail485
  baud_rate: 9600
  parity: EVEN
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ' ');

external_components:
  - source: github://tristone-cz/esphome_samsung_hvac_bus@patch-slow-registration  
    components: [samsung_ac]
    refresh: 0s

samsung_ac:
  non_nasa_keepalive: true
  debug_log_undefined_messages: true
  debug_log_messages: true
  capabilities: 
    vertical_swing: true
    presets: 
      quiet: true
      fast: true
      windfree: true

  devices:
    - address: "c8" # Outdoor device address
      outdoor_temperature: 
        name: "Outdoor temperature"
      outdoor_instantaneous_power:
        name: "Outdoor Instantaneous Power"
      outdoor_cumulative_energy:
        name: "Outdoor Cumulative Energy"
      outdoor_current:
        name: "Outdoor Current"
      outdoor_voltage:
        name: "Outdoor Voltage"
    - address: "00" # Indoor device address 
      capabilities:
        horizontal_swing: false  
      climate:
        name: "Bedroom AC"
      room_temperature:
        name: "Bedroom AC temperature"
      target_temperature:
        name: "Bedroom AC target"
      power:
        name: "Bedroom AC power"
      mode:
        name: "Bedroom AC mode"
      room_temperature_offset: 0
      error_code:
        name: "Error Code"
    - address: "01" # Indoor device address 
      capabilities:
        horizontal_swing: false  
      climate:
        name: "Kids AC"
      room_temperature:
        name: "Kids AC temperature"
      target_temperature:
        name: "Kids AC target"
      power:
        name: "Kids AC power"
      mode:
        name: "Kids AC mode"
      room_temperature_offset: 0
      error_code:
        name: "Error Code"
```

wifi75, 2025-04-03:
my board is like this below...

 https://shop.m5stack.com/products/atom-lite-esp32-development-kit?srsltid=AfmBOopko5szsdUMkcIXOMMgOof-kGve9twSYXNWtqKwDwqtCw3bCyYf

wifi75, 2025-04-03:
> ```
> esphome:
>   name: ac-rooms
>   friendly_name: ac-rooms
>
> esp32:
>   board: esp32-s3-devkitc-1
>   flash_size: 8MB
> #  framework:
> #    type: esp-idf
>
> web_server:
>   port: 80
>
> logger:
>   level: INFO [...]

so I managed to get something, now there is communication, at least in reception, in fact if I use the remote control to turn on or change temperature, on HA I get the variations, but if I try to change temperature or turn on or off from HA it does not execute the command on the climate.

what could it be?

tristone-cz, 2025-04-04:
Can you open the web dashboard of the ESP? In ESPhome there is button VISIT.
Do you see there the updates from the the remote? Eg. target temperature is changing?

Cause one thing is that you can see the bus traffic another one if the EPS is registered to the communication and receives the data as a target entity.