omerfaruk-aran/esphome_samsung_hvac_bus #192: Incompatibility with esphome 2024.10.0

kst84, 2024-10-19:
Hi, It looks like the code is not compatible with recent release of esphome 2024.10.0

`NFO ESPHome 2024.10.0
INFO Reading configuration /config/klimatyzacja-gora.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1018, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1005, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 485, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 206, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 186, in wrapped
    await coro(conf)
  File "/config/.esphome/external_components/11b80f5a/components/samsung_ac/__init__.py", line 304, in to_code
    preset_conf.get(CONF_PRESET_NAME, preset_info["displayName"]),
    ^^^^^^^^^^^^^^^
AttributeError: 'bool' object has no attribute 'get'`

omerfaruk-aran, 2024-10-19:
The error you're encountering:

``` 
AttributeError: 'bool' object has no attribute 'get' 
```

is likely caused by a configuration issue in your YAML file, specifically in the section where you define presets. Based on the traceback, it seems that the code is expecting a dictionary (which uses the `.get()` method) but instead, it received a boolean value (`true` or `false`). This error commonly occurs when the configuration of `presets` is not properly formatted.

Here's the issue breakdown:

In your configuration file under` capabilities -> presets`, you have:

```
presets: 
  # short version
  quiet: true
  # long version - allows to localize names
  quiet:
    name: "Makes no sound"
    enabled: true
```

The problem here is that you're defining `quiet `twice — once as `true `(a boolean), and again as a dictionary (with `name `and `enabled`). The YAML parser gets confused because it can't decide which form to use, and this is likely leading to the `bool object` error you're seeing.

### Suggested Fix:
Choose only one way `quiet `pre

```
presets: 
  quiet: true
```

Or

```
presets: 
  quiet:
    name: "Makes no sound"
    enabled: true
```

Please

kst84, 2024-10-19:
@omerfaruk-aran it's not related to quiet mode. I have it configured properly.
The error states something wrong with preset_info["displayName"]).

omerfaruk-aran, 2024-10-19:
@kst84 Can you send your full yaml example?

kst84, 2024-10-19:
Sure
`esphome:
  name: samsung-ac-gora
  friendly_name: Klimatyzacja

esp32:
  board: m5stack-atom

web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR 
api:

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid_lazy
  password: !secret wifi_password_lazy

captive_portal:

switch:
  - platform: restart
    name: "Restart"

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

samsung_ac:
  capabilities: 
    vertical_swing: true
    horizontal_swing: true
    presets: 
      quiet: true
      windfree: true
      fast: true

  devices:
    - address: "20.00.00"
      climate:
        name: "Sypialnia climate"
      room_temperature:
        name: "Sypialnia temperature"
      target_temperature:
        name: "Sypialnia target temperature"
      power:
        name: "Sypialnia power"
      mode:
        name: "Sypialnia mode"
      room_humidity:
        name: "Sypialnia humidity"
      automatic_cleaning:
        name: "Sypialnia automatic clean"

    - address: "20.00.01"
      climate:
        name: "Pokoj climate"
      room_temperature:
        name: "Pokoj temperature"
      target_temperature:
        name: "Pokoj target temperature"
      power:
        name: "Pokoj power"
      mode:
        name: "Pokoj mode"
      room_humidity:
        name: "Pokoj humidity"
      automatic_cleaning:
        name: "Pokoj automatic clean"

    - address: "10.00.00"
      outdoor_temperature:
        name: "Outdoor temperature"
      custom_sensor:
        - name: Power
          message: 0x8413
          device_class: power
          state_class: measurement
          unit_of_measurement: W
        - name: Energy
          message: 0x8414
          device_class: energy
          state_class: total_increasing
          unit_of_measurement: kWh
          accuracy_decimals: 2
          filters:
            - multiply: 0.001
`

omerfaruk-aran, 2024-10-19:
OK, I can see that it is used incorrectly in your yaml example. 

Could you edit your code 

```
samsung_ac:
  capabilities: 
    vertical_swing: true
    horizontal_swing: true
    presets: 
      quiet: true
      windfree: true
      fast: true
```

to

```
samsung_ac:
 capabilities:
   vertical_swing: true
   horizontal_swing: true
    presets:
      quiet:
        name: "Makes no sound"
        enabled: true
      fast:
        name: "Fast cooling"
        enabled: true
      windfree:
        name: "Wind-Free mode"
        enabled: true
```

change this and try it?

kst84, 2024-10-19:
@omerfaruk-aran yeah it compiles now. The question is how it comes to example.yaml ?

https://github.com/lanwin/esphome_samsung_ac/blob/main/example.yaml#L75

omerfaruk-aran, 2024-10-19:
Hi, you're absolutely right! The example YAML was previously missing the flexibility to handle both the short and long versions of the presets. We've now made adjustments to allow users to configure presets using either format, depending on their preference.

You can now use either:
for a quick configuration 
```yaml
quiet: true
```
or for a more detailed setup.
```yaml
quiet:
  name: "Makes no sound"
  enabled: true
```
or mixed
```yaml
quiet:
  name: "Makes no sound"
  enabled: true
 fast: true
```
We've also updated the example.yaml to reflect this change. You can check out the updated PR here: #193 

Thank you for pointing this out, and feel free to test it out!

kst84, 2024-10-20:
@omerfaruk-aran Thank you for your help and clarification on the above. Cheers.