omerfaruk-aran/esphome_samsung_hvac_bus #195: MQTT issue

Digy-dp, 2024-10-20:
Hi, I’m having a problem with the configuration.
The first difference is that I’m using the M5Stack AtomLite S3 (white) board, but the issue itself relates to MQTT.
Here is the error log:

src/esphome/components/samsung_ac/debug_mqtt.cpp: In function 'void esphome::samsung_ac::debug_mqtt_connect(const string&, uint16_t, const string&, const string&)':
src/esphome/components/samsung_ac/debug_mqtt.cpp:49:26: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'host'
   49 |                 mqtt_cfg.host = host.c_str();
      |                          ^~~~
src/esphome/components/samsung_ac/debug_mqtt.cpp:50:26: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'port'
   50 |                 mqtt_cfg.port = port;
      |                          ^~~~
src/esphome/components/samsung_ac/debug_mqtt.cpp:53:30: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'username'
   53 |                     mqtt_cfg.username = username.c_str();
      |                              ^~~~~~~~
src/esphome/components/samsung_ac/debug_mqtt.cpp:54:30: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'password'
   54 |                     mqtt_cfg.password = password.c_str();
      |                              ^~~~~~~~
*** [.pioenvs/samsung-pc/src/esphome/components/samsung_ac/debug_mqtt.o] Error 1
========================= [FAILED] Took 118.24 seconds =========================

omerfaruk-aran, 2024-10-20:
It seems like you are encountering an issue related to the MQTT configuration in your samsung_ac component.

To better assist you, could you please share the YAML configuration file you’re using? This will help us understand the exact setup and provide more tailored support.

In your response, please include:

	•	The version of ESPHome you are using
	•	The Samsung AC integration configuration
	•	The MQTT configuration in your YAML file
	•	Any other relevant information that could help replicate the issue

Once we have your YAML file, we can further investigate and provide a solution. Thank you!

Digy-dp, 2024-10-20:
Thank you for response, i'll sent details:

1. ESP Current version: 2024.10.0
2. yaml file:
`esphome:
  name: samsung-pc
  friendly_name: Samsung_PC
esp32:
    board: m5stack-atoms3
    variant: esp32s3
    framework:
      type: esp-idf
      version: 5.0.2
      platform_version: 6.3.2
web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR

api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""
# Setup your WiFi password
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  tx_pin: GPIO6
  rx_pin: GPIO5
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

samsung_ac:

  debug_mqtt_host: !secret mqtt_broker
  debug_mqtt_port: 1883
  debug_mqtt_username: !secret mqtt_username
  debug_mqtt_password: !secret mqtt_password
  
  debug_log_undefined_messages: false
  
  debug_log_messages: false
 
  devices:`

omerfaruk-aran, 2024-10-20:
To resolve the issue, please update your configuration by adding the following line under the external_components section. This will ensure the component is refreshed and reloaded with the latest version:

```yaml
external_components:
  - source: github://lanwin/esphome_samsung_ac@main 
    components: [samsung_ac]
    refresh: 0s
```
After making this change, reload your YAML file and try running the update again. Let us know if the issue persists.

Digy-dp, 2024-10-20:
Still the same problem:
[image]

Code:

```
external_components:
  - source: github://lanwin/esphome_samsung_ac@main 
    components: [samsung_ac]
    refresh: 0s

samsung_ac:

  debug_log_undefined_messages: false 
  debug_log_messages: false

  devices:
```

omerfaruk-aran, 2024-10-20:
Please check the following steps to ensure your setup is correct:

### 1. Verify secrets.yaml Configuration:
- Open your `esphome>secrets.yaml` file and check that `mqtt_broker`, `mqtt_username`, and `mqtt_password` are correctly defined and match the configuration you're using in your YAML file.
- Ensure there are no typos or missing fields in the `secrets.yaml` file.

### 2. Ensure MQTT is Running on Home Assistant:
- Confirm that the MQTT broker is installed, properly configured, and running on your Home Assistant instance.
- You can verify this by checking the MQTT integration in Home Assistant and ensuring it is correctly publishing and receiving messages.

### 3. Test MQTT Broker Connectivity:
- If possible, use an external MQTT client (like MQTT Explorer) to connect to your MQTT broker using the same credentials from your `secrets.yaml` to confirm the broker is operational.

### 5. Same Setup Working for Others:
- Please note that we have tested the same ESPHome, Samsung_AC component, and MQTT setup with the same ESP-IDF version, and it is working without any issues. This indicates that the problem may be specific to your environment or configuration.
- After verifying the above steps, try to compile and upload the firmware again. Let us know if the issue persists, and we can assist further!

omerfaruk-aran, 2024-10-21:
Hi @Digy-dp,

I just wanted to check in and see if the issue you were experiencing with MQTT has been resolved. If everything is working now, could you please close the issue? If you're still having trouble, feel free to let us know, and we'll continue to assist you.

Thanks for your collaboration!

Best regards!

closed 2024-10-23 by Digy-dp