omerfaruk-aran/esphome_samsung_hvac_bus #214: [NonNASA] Outdoor Temperature and Energy Consumption - Home Assistant Sensor Request

omerfaruk-aran, 2024-10-29:
### Discussed in https://github.com/omerfaruk-aran/esphome_samsung_hvac_bus/discussions/108

<div type='discussions-op-text'>

<sup>Originally posted by **Goaheadz** March  6, 2024</sup>
Since there is already a message containing the outdoor temperature, I think making this as a HA sensor, would be handy for automation's.
This is the message I get:

 [W] | [samsung_ac:493] | MSG: {src:c8;dst:00;cmd:c0;commandC0:{ou_operation_mode:88; ou_4way_valve:0; ou_hot_gas_bypass:0; ou_compressor:0; ou_ac_fan:0; ou_outdoor_temp[°C]:13; ou_discharge_temp[°C]:11; ou_condenser_mid_temp[°C]:12}}
 
 Let me know what can I do, to make this possible.
 
 Help needed :)
 
 Thx
 G

</body>
</html></div>

> Beside de temperature, there is also the voltage and amperage being draw by the outdoor machine. Since power from all splitters are connected to the outdoor, we could also use this information as sensors for consuption log and display.
> 
> like adding a new power draw like aqs in picture:
> 
> ![image](https://private-user-images.githubusercontent.com/13796554/323778988-4b994895-6756-4660-8177-9e4ba493a709.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxOTcxMzksIm5iZiI6MTczMDE5NjgzOSwicGF0aCI6Ii8xMzc5NjU1NC8zMjM3Nzg5ODgtNGI5OTQ4OTUtNjc1Ni00NjYwLTgxNzctOWU0YmE0OTNhNzA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI5VDEwMTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4MmI5ZmU0NGNiY2YzOTgzMjQ4MWQxYzViMGUxN2VmMGUwMmZhY2VkMmMyYzkxNDMzMTcwNTk4ZTlhY2Q5NWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qK-uyVlPRv5rnEv5_OQoNpKWMcXFcWOc_IoJWIpOb8E)

> Hi Aran,
> 
> Thank you so much for all the time you put into this.
> 
> As soon as I get home (later in the day) I will also provide some log files and yaml (currently the AC breaker is OFF), and try the new sensor.
> 
> devices: - address: "c8" # Outdoor device address outdoor_temperature: name: "Outdoor temperature"
> 
> Adding inverter_power[W] sensor, will also allow us, to create a Energy sensor in Home Assistant and control consumption and costs per outdoor unit.

Goaheadz, 2024-10-29:
Hi Aran,

Like said, I will test this later in the day, but in the mean time, and to better understand the good work you are doing:

-  Is this the "inverter_power[W]" reported in ESP log files?
    outdoor_instantaneous_power:
      name: "Outdoor Instantaneous Power"
  
-  Not sure if outdoor unit reports VAC, but it seems to report VDC "inverter_voltage[VDC]" at least for DC invert units.
Is this sensor VAC or VDC?
    outdoor_voltage:
      name: "Outdoor Voltage"    

- Not sure if outdoor unit reports Energy. usually this is done inside Home Assistant with new energy sensor, using Volts and Watts in time (Wh) 
What does this sensor reports?
    outdoor_cumulative_energy:
      name: "Outdoor Cumulative Energy"
      
- Not sure if outdoor unit reports AAC (Amps in AC side), cause logs say "inverter_current[ADC]"
If we have instant Watts and VAC we can calculate all othe parameters (Wh, current in AC, etc)
Does this sensor resports amps in AC or DC?
    outdoor_current:
      name: "Outdoor Current"

Sorry for so many questions. 

Goahead

tech2urdoor, 2024-10-29:
Hi Omer,

You are a superstar! I haven't tested everything completely yet but all the sensors are coming up in Home Assistant! Thank you so much for implementing this. You should be proud. 

Brad

omerfaruk-aran, 2024-10-29:
Hi Brad,

Thank you so much for your kind words! 😊 I'm really glad to hear that all the sensors are showing up in Home Assistant. Your feedback is invaluable, and it's great to know the implementation is working well so far.

If you could continue testing and let me know if everything remains stable or if you encounter any issues, that would be super helpful. Once you're confident that all is working as expected, I'll be happy to merge the changes from the testing source to the main branch.

Looking forward to your feedback!

Best,
Ömer Faruk ARAN

Jkirjo, 2024-10-30:
Here is now so lot of messages that i can't keep count what is relevant and which not...
All currently implemented works fine, which is very fine.
But there is few glitch in documents/messages. In external components part that [samsung_hvac_nasa] is invalid. The source code not contain that reference. When keep the previous [samsung_ac] in there, it works fine.
Also the new -address "c8" is unnecessary, outside temperature works fine under normal outdoor unit address. As power and consumption readings too.
At the end, i have a wish to make. Could you please add every available sensor value to source code? Then user could select in yaml configuration, what he/she want to see. As we see in this thread start, there is many values reported in bus messages.

omerfaruk-aran, 2024-10-30:
Hello,

First of all, thank you for your feedback! The term "samsung_hvac_nasa" was actually the name of my test system, and it accidentally made its way into the documentation during the discussion. Apologies for the confusion, and this will be corrected.

Regarding the C8 topic: We specifically set this address for adding the sensors under the outdoor unit. If these sensors were added to each indoor unit, it would lead to duplicated sensors. Therefore, we chose this approach to improve efficiency and prevent data redundancy.

When you mention all available sensors, could you clarify which specific sensors you are referring to? As we continue to decode the system, we are adding new sensors gradually. If you are capturing additional sensor logs or values, please feel free to share them with us. This will help us include any missing or newly discovered sensors.

Jkirjo, 2024-10-30:
> When you mention all available sensors, could you clarify which specific sensors you are referring to? As we continue to decode the system, we are adding new sensors gradually. If you are capturing [...]

In this thread start message is some log view, where is undefined sensor values. If they are available in bus messages, why discard them... These would be useful sometimes.

> W] | [samsung_ac:493] | MSG: {src:c8;dst:00;cmd:c0;commandC0:{ou_operation_mode:88; ou_4way_valve:0; ou_hot_gas_bypass:0; ou_compressor:0; ou_ac_fan:0; ou_outdoor_temp[°C]:13; ou_discharge_temp[°C]:11; ou_condenser_mid_temp[°C]:12}}

tech2urdoor, 2024-10-30:
Hey Omer,

I will have to wait till I get home to send you the logs but I turned on my AC remotely today and the voltage and instantaneous sensors aren't updating. They still say 0. Outdoor temperature is working perfectly and the current sensor is updating but seems to be incorrect. The cumulative energy sensor is updating but is obviously incorrect as well.  Just wanted to let you know. I'll share my log with you later. Jkirjo: I think those undefined values are pretty useless info. I wouldn't want him to waste his time with those personally but perhaps someone would have a use for them. 

Thanks
Brad

tech2urdoor, 2024-10-30:
[image]
[atom_yaml.txt](https://github.com/user-attachments/files/17568463/atom_yaml.txt)
[logs_atom_logs.txt](https://github.com/user-attachments/files/17568464/logs_atom_logs.txt)

omerfaruk-aran, 2024-10-30:
Hello Brad,

Thank you for the update! From the logs I reviewed earlier, it seems that the energy values are indeed showing as 0, which explains why the sensors are not updating correctly. To get to the bottom of this, I’ll need more detailed logs. If there are others with different models who can share their log information, it would help us troubleshoot and test potential fixes more quickly.

I appreciate your patience, and please do share your logs when you have a chance. This additional information will be crucial in identifying the root cause and resolving the issue effectively.

Thanks,
Ömer Faruk ARAN

omerfaruk-aran, 2024-10-30:
Hello Brad,

To help narrow down the issue, could you let me know which pins you connected your air conditioner to (e.g., F1/F2/F3/F4)? Also, could you specify which ESP device and RS485 module you’re using in your setup?

This information will be really helpful in understanding the configuration and diagnosing the problem accurately.

Thanks,
Ömer Faruk ARAN

Goaheadz, 2024-10-30:
Hi Brad,

"I turned on my AC remotely today and the voltage and instantaneous sensors aren't updating. "

- You gave power to the outdoor unit or did you turn on 1 indoor unit?
- Did you wait at least 2-3 minutes for the outdoor unit ends the startup procedure and after that the indoor unit starts requesting temperature to the outdoor?
- Did the request you made to the indoor unit, using the remote, would trigger work to the outdoor unit? (big temperature difference request)

We are talking about VDC (outdoor DC inverter unit), so I would assume that only if the compressor starts working/ramping up, till then, there will be no VDC values ...just guessing.

In your log file its seems that the outdoor unit did not started the compressor yet.
"inverter_current[ADC]:0.800000; inverter_voltage[VDC]:0.000000; inverter_power[W]:0.000000"

I would also guess that 0.8A is the current being draw for the startup procedure. (valves , electronics, sensors, relays, outside FAN etc etc), cause its just +/-180W of consumption in DC side. (depending in efficiency of the inverter, that would be +/-200W in AC side)..at this point the compressor did not started, so inverter_power[W] is also 0

I believe that if you use FAN mode only, the result would be the same as above in log files. (I need to test this also)
Just be sure that the compressor is working (you can also see that in frequency values changes in log files)...give it some time.

At least, that's how I understand the system works.

I was not able to test this yesterday (got home pretty late), but I will try to test it today and give you all some reports to.

Regards
G.

omerfaruk-aran, 2024-10-30:
> I will take a more detailed log while running it later tomorrow (it's a nice temperature right now). I presume you would want more detailed logs while it is running and generating different values. It is connected to the F1 and F2 connections. I purchased the 'M5 Stack Official ATOM Lite ESP32 Development Kit' along with the 'M5Stack Official ATOMIC RS485 Base TTL-RS485 Converter'. At least that's what they were detailed as from AliExpress. They look exactly like the items that were recommended. No need to thank me for my patience. You are the one doing the hard work! I am very thankful and quite happy to provide any info you need. I imagine it wouldn't be easy when you don't have the device yourself to test.

Hello Brad,

Thank you for the details and for your support! Yes, having detailed logs while the AC is running and generating different values will be quite helpful.

Just to confirm, in your YAML configuration, have you set the baud_rate to 2400? If not, could you please try setting it to 2400 and let me know if that makes any difference? The F1 and F2 connections should be correct, but the baud rate setting can sometimes affect communication.

I really appreciate your cooperation. We’re making good progress thanks to your help!

Best regards,
Omer Faruk ARAN

omerfaruk-aran, 2024-10-30:
Hello everyone,

I have made significant code optimizations for the Non-NASA implementation. These changes include improved memory management and substantial updates to the logging structure to log all data comprehensively. If a few of you could help test this latest version, it would be greatly appreciated. Based on the results, I can plan the next steps accordingly.

For those who will be testing, please share YAML examples, log dumps, and device connection information with me.

```yaml
external_components:
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@add-outdoor-sensors 
    components: [samsung_ac]
    refresh: 0s
```

Thank you!

Goaheadz, 2024-10-30:
I will do this later in the evening for sure.
Thanks again for your hard work so far.

Goaheadz, 2024-10-30:
Aran,

Just changing in YAML:

`external_components:
#  - source: github://lanwin/esphome_samsung_ac@main
#    components: [samsung_ac]  
#    refresh: 0s
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@add-outdoor-sensors 
    components: [samsung_ac]
    refresh: 0s`
    
  ....And adding the outdoor sensors in YMAL, nothing works  anymore. Integration broken or maybe log format not correct, so that integration can see values.
   
[logs_samsung-hvac-muti-split_run (2).txt](https://github.com/user-attachments/files/17578206/logs_samsung-hvac-muti-split_run.2.txt)

My YAML:
[samsung-hvac-muti-split.yaml.txt](https://github.com/user-attachments/files/17578536/samsung-hvac-muti-split.yaml.txt)

Goaheadz, 2024-10-30:
with UART debug:

[logs_samsung-hvac-muti-split_run (4).txt](https://github.com/user-attachments/files/17578640/logs_samsung-hvac-muti-split_run.4.txt)

Goaheadz, 2024-10-30:
[image]

omerfaruk-aran, 2024-10-30:
Could you please change the baud rate to 2400, re-upload the configuration, and test again? If you have the time, I can also connect to your computer via Anydesk to perform the tests myself. Feel free to reach out to me on Instagram at @teknoaran for further assistance.

Goaheadz, 2024-10-30:
2400? Why?
If everything is working with 9600 with @lanwin code, it makes no sense to change it just for outdoor sensors..

omerfaruk-aran, 2024-10-30:
The reason for setting the baud rate to 2400 is that non-NASA protocol devices specifically communicate using a 2400 baud rate, according to the protocol specification. While the 9600 baud rate works for NASA protocol devices, the non-NASA devices require 2400 baud for proper communication, including outdoor sensor data. Without this change, the integration won't be able to read or send data correctly to these non-NASA devices.

The main purpose of this test is to confirm the baud rate configuration. In previous attempts, the outdoor temperature and energy sensors were receiving incorrect values, which caused the sensors to not update correctly. If the baud rate adjustment is correct, this change should resolve that issue, allowing the sensors to receive accurate data.

Goaheadz, 2024-10-30:
I still dont get it...sorry :)
If I can ,for 3 splits,  turn AC on, off, change temperature , see indoor ambient temperature for each split etc etc with 9600, why would I change that for the outdoor data.
My AC is non-Nasa and the correct baud rate is 9600. Maybe the protocol changed along the years.

In the past, when I started debuging with @lanwin I was able to see this data format in logs like this:

[W][samsung_ac:543]: MSG: {src:c8;dst:00;cmd:f3;commandF3:{inverter_max_frequency[Hz]:66; inverter_total_capacity_requirement[kW]:7.700000; inverter_current[ADC]:2.400000; inverter_voltage[VDC]:0.000000; inverter_power[W]:0.000000}}

Today I dont remember how to enable this type of raw data in yaml, and thats probably why it doesnt work for me.
I need to enable this for your code be able to catch values.

omerfaruk-aran, 2024-10-30:
Yes, the data captured by my code at a speed of 2400 baud is very important, I will direct the project accordingly. The settings you have changed now are for testing, we will restore it after the test.

omerfaruk-aran, 2024-10-30:
I removed the small data size control that I added for non-NASA purposes. While the baud_rate is 9600, can you update it and try again?

Goaheadz, 2024-10-30:
[image]
[image]
[image]

Goaheadz, 2024-10-30:
Ok.
Now we have data.

[samsung-hvac-muti-split.yaml.txt](https://github.com/user-attachments/files/17579882/samsung-hvac-muti-split.yaml.txt)

omerfaruk-aran, 2024-10-30:
Great, @Goaheadz! Getting the voltage correctly was crucial. Now, let’s see if we can observe the Current, Power, and Cumulative Power values by running one of the air conditioners.

tech2urdoor, 2024-10-30:
Just to add to all this. This morning I quickly changed the baud rate and it broke the whole integration. Nothing was working. I changed the baud rate back to what it was and it was still broken. So I had to change back to the main one just so I can control it remotely again. I'll update you with my logs and yaml when I get home.

omerfaruk-aran, 2024-10-30:
Hello @tech2urdoor, we updated it 2 hours ago, if you update it again while the baud rate is at 9600, there should be no problem.

Goaheadz, 2024-10-30:
[image]
 Data is changing and status also:
 
[image]

 
 When I press in HA to turn AC off or On it does not work and integration becomes unavailble for some seconds till it recovers again.
 
 
[image]
.

omerfaruk-aran, 2024-10-30:
@Goaheadz I understand, I will share an update for that problem now, can you wait a bit? It won't take 10 minutes.

Goaheadz, 2024-10-30:
If  press off, in the remote, the issue is the same.
Integration becomes unknown for a few seconds and then recovers data again.

omerfaruk-aran, 2024-10-30:
OK, I updated, can we install it again and test it?

Goaheadz, 2024-10-30:
[logs_samsung-hvac-muti-split_run (5).txt](https://github.com/user-attachments/files/17580026/logs_samsung-hvac-muti-split_run.5.txt)

It kind of works but its slow response and sometimes it doesnt respond to command.
I think there is to much data flowing and esp8266 and ac unit cannot catch commands. (errors)
In log file you can see the data error flood.

Goaheadz, 2024-10-30:
[image]
[image]

omerfaruk-aran, 2024-10-30:
Yes, I did this on purpose. There are a lot of undefined sensors and actions for non-NASA. I added a few logs to capture these, but one of them was duplicated, so I removed it now. I sent an update, can we do a final test again? It would be great if you could send the log again. :) You can take a few actions before sending.

Goaheadz, 2024-10-30:
Let me know when I can compile again.

[image]

omerfaruk-aran, 2024-10-30:
There is no problem but can you try again?

Goaheadz, 2024-10-31:
Still very slow response and fails sometimes when I press buttons in HA.

suddently everything fails and recovers some minutes after

[image]

I do think esp8266 is getting to much information and laging alot

omerfaruk-aran, 2024-10-31:
Okay, I turned off the extra logs and sent an update, it should arrive within 1 minute. By the way, I ask you, can you write the models of the outdoor and indoor units?

Goaheadz, 2024-10-31:
no problem..I will get the partnumbers tomorrow, as its already late, night and cold outside eheheh

omerfaruk-aran, 2024-10-31:
ok :) thank you for your support. I have an update planned for ESP8266 soon, hopefully it will solve the memory problem. I'm going to bed too.

Goaheadz, 2024-10-31:
Thank you for all your time you spend into this.
See you tomorrow

Jkirjo, 2024-10-31:
> no problem..I will get the partnumbers tomorrow, as its already late, night and cold outside eheheh

With interest i follow this also. Regardless that i have nasa-device.
Winter is coming, here is now below freeze outside.

tech2urdoor, 2024-10-31:
Hi Omer,

I've attached a good 5+ minutes of logs. I turned it on through Home Assistant and turned the temp down to 19, so the compressor was running. All the controls worked fine, and all the sensors were available, but the voltage and instantaneous power sensors never updated, and the current sensor was updating but incorrect, as was the cumulative sensor. I have a CT clamp on the AC circuit so I can already monitor it, I just want everyone to get the benefit of that data. We are in different time zones but I am quite happy to offer my computer up for you to remotely login if needed and if our free time matches for testing.

[logs_atom_run.txt](https://github.com/user-attachments/files/17582477/logs_atom_run.txt)

[image]

Goaheadz, 2024-10-31:
3x Splits - AQV18PSBN
[image]
[image]

1x Outdoor - AJ100 or RJ100 or FJM100 (have to recheck the correct model)
Main Board:
[image]

omerfaruk-aran, 2024-10-31:
@tech2urdoor can you write the models of the outdoor and indoor units?

tech2urdoor, 2024-10-31:
It is a ducted system with one outdoor unit: AC090HCAFKH/SA I'm not sure what model the 'Indoor' part of it would be as that's located in the ceiling space, if that makes sense. There is a wired remote control: MWR-WE10 

There is a completely separate zonemaster controller that controls dampeners (turning vents in rooms on and off) but I believe that is not directly connected at all as even the official wifi kits for the AC couldn't control zones.

omerfaruk-aran, 2024-11-01:
Hey Everyone,

I’ve made an important update for the Non-NASA protocol. This update introduces a dynamic delay mechanism during the startup phase. Now, the controller registration requests are adjusted to ensure they’re sent only after the outdoor unit is fully ready. The system will start with a 10-second delay and adapt up to 1 minute, providing more reliable and consistent communication.

Key Focus for Testing:

Energy Sensors: Please pay close attention to the energy sensors (current, power, voltage, and cumulative energy) and let me know if the values are now accurate and updating correctly.
ESP8266 Devices: If you’re using ESP8266-based devices like the D1 Mini, watch for any performance improvements, such as reduced lag or better memory management.
How to Help:

Update your code and upload it to your device.
Carefully monitor behavior, especially the energy sensor readings, and note any changes or issues.
If you notice any problems or improvements, please share your feedback and log outputs.
Your feedback is extremely valuable and will help me continue improving this project. Thanks for testing and for your support!

tech2urdoor, 2024-11-01:
Hi Omer,

I just updated and uploaded the code. I switched it on through home assistant and lowered the target temperature so the compressor would run. Everything worked fine with no lag but unfortunately, the voltage and instantaneous power sensors never updated in home assistant. Same experience as last time. 

Thanks
Brad
[logs_atom_run.txt](https://github.com/user-attachments/files/17595790/logs_atom_run.txt)

omerfaruk-aran, 2024-11-01:
@tech2urdoor Can you update it again and run another test?

tech2urdoor, 2024-11-01:
Same result. Logs attached
[logs_atom_run.txt](https://github.com/user-attachments/files/17597140/logs_atom_run.txt)

jszkiela72, 2024-11-01:
Hi Omer,

First I want to thank you for your work on the development of the Non-NASA Samsung AC system.

My system is:

Outdoor Unit
AJ100FCJ5EH/EU
Indoor unit
4x AR09HSFSBWKNZE
1x AR12HSSDBWKNEU

For integration I use m5stack-atom with baud_rate: 9600. Communication works fine.
In HA I see all new sensors.
However, it seems to me that the current of the outdoor unit is underestimated by 10 times. Can you correct this?

[image]
[Samsung AC Non-NASA system.yml.txt](https://github.com/user-attachments/files/17599652/Samsung.AC.Non-NASA.system.yml.txt)

omerfaruk-aran, 2024-11-01:
Hi @jszkiela72,

Thank you for your valuable feedback and for providing detailed information about your setup. It's possible that the current sensor value for the outdoor unit is indeed off by a factor of 10, likely due to a calibration issue. I’ll look into this and work on a correction for accurate current readings. I’ll update you soon with a potential fix.

omerfaruk-aran, 2024-11-01:
Hi @tech2urdoor,

Based on my research, there can be issues with sensor data not being read correctly due to differences in some air conditioner models, as well as the ESP devices and RS485 modules used. Specifically, problems may arise when the baud rate configurations vary. Although a baud rate of 9600 might work for some devices, it could be necessary to test other baud rate speeds. Could you conduct more extensive testing to see how these different configurations affect the results? For example, could you try changing the baud rate to 2400, 4800, and 9600, and then create separate log records for each, and send them?

Thank you, and best of luck!

Goaheadz, 2024-11-01:
I also confirm the 10x off in current:
W=VxA is not valid :)
[image]

I notice improvements but still to much lag and fail in requests.
(I can notice lagg cause AC beeps everytime we change parameters..if it doesnt beep, then message was not sent or was not received.)
- Power on had a fast response for first unit, but the second one took +10seconds.
- After 2 splits getting on, changing one of them from fan to heat (per example) , no beeps received.
- After pressing several times the "change mode from fan to heat" beep came.

Its hard to explain behavior, cause I can see sometimes the integration change to "unavailable" for brief second, if i press  from an off state to "fan" state per example. After that if I press again "fan", it will start (beep), and sometimes it responds quikly.

My 2 cents would be, something flooding the D1 mini.
I remember having this in initial tests with Lanwin, but I dont know what changes he made to solve the lagg.

In log files:
Received unknown command:
C1, F0, F1 from C8 to all internal splits.

[logs_samsung-hvac-muti-split_run (7).txt](https://github.com/user-attachments/files/17604021/logs_samsung-hvac-muti-split_run.7.txt)

Keep up the good work.

Goaheadz, 2024-11-01:
[samsung-hvac-muti-split.yaml (2).txt](https://github.com/user-attachments/files/17604052/samsung-hvac-muti-split.yaml.2.txt)

Should I remove some enable logs in yaml code to easy the messages?

tech2urdoor, 2024-11-01:
Unfortunately if I use any other baud rate other than 9600 everything becomes unavailable in Home Assistant. Attached logs for each baud rate.

Thanks
[logs_atom_run_9600.txt](https://github.com/user-attachments/files/17604769/logs_atom_run_9600.txt)
[logs_atom_run_4800.txt](https://github.com/user-attachments/files/17604770/logs_atom_run_4800.txt)
[logs_atom_run_2400.txt](https://github.com/user-attachments/files/17604771/logs_atom_run_2400.txt)

htht2001, 2024-11-03:
Do the energy sensors support more than one outdoor unit?
I have two outdoor units.
[image]
By the way, is error_code still active for the indoor units, in the indoor units it appears as NA
And while in the external units it appears as 0
[image]
(L3 is a facial unit and 00 and 01 are outdoor units)

If there is anything I can help with
I will gladly help
Thanks again for the amazing project

tristone-cz, 2024-11-05:
Hi guys.

So finally I tried as well.
In general it seems to work, getting the values.

But few interesting things and even limitations.

First, the power consumption is obviously related only to the outdoor inverter itself. I have metered socket for the AC and I can see the consumption. Which is few watts in total doing nothing or about 20 W if fan is running but not cooling / heating using the inverter. And in such cases the ESP returns 0 consumption.

Second, the measurement values are somehow strange.
Voltage - do you know if the inverter is somehow transforms the input voltage? Cause it shows 310-350V but in the socket here is 230 V.
Consumption - it shows much higher consumption compared to my metered socket:
```
[11:03:42][W][samsung_ac:565]: MSG: {src:c8;dst:01;cmd:f3;commandF3:{inverter_max_frequency[Hz]:60; inverter_total_capacity_requirement[kW]:0.000000; inverter_current[ADC]:1.500000; inverter_voltage[VDC]:310.000000; inverter_power[W]:465.000000}}
```
[image]
465 W reported by outdoor unit vs. 273 W reported by the socket.

tech2urdoor, 2024-11-05:
I'm jealous! I'm not getting any readings in home assistant for voltage or instantaneous power even though that data is being communicated (according to the logs). I also have an independent reading from a Shelly em with a ct clamp over the AC circuit so this isn't a necessity for me. The whole circuit actually draws around 15w with AC off. I'm guessing that is power that it uses to power the control panel and the m5stack chip that is wired into that circuit. Although having this working would allow me to use that ct clamp to monitor a different circuit. I'm surprised your AC is actually connected to a socket. Here in Australia, all high powered devices like air conditioners have to be on an independent circuit. They draw too much power for a standard socket.

[12 more replies]

closed 2026-01-02 by omerfaruk-aran