omerfaruk-aran/esphome_samsung_hvac_bus #221: `Bug` [Samsung_HVAC]: HVAC does not start until the second try

mbo18, 2024-11-06:
### The problem

I have this issue on all my units: when it has been off for several hours, when I set the HVAC from off to heat or cold, the unit does not go on. It stay off and I need to do it a second time.

Note that since the latest update to the code, this issue is only when I need to start the unit. When on, all commands are correctly executed on the first try (thank you for this!)

I will try to get logs and will update this issue

### ESPHome Version

2024.10.2

### Samsung HVAC Component Version

Latest from main

### Home Assistant Version

2024.10.4

### AC Type

NASA

### Indoor AC Unit Model

AR09TXCAAWKN

### Outdoor AC Unit Model

AR09TXCAAWKX

### ESP Device Model

M5stack atom lite + m5stack rs485

### Connection Points

F1/F2

### YAML Configuration

```yaml
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@main
    components: [samsung_ac]

samsung_ac:
  debug_log_messages_raw: false
  debug_log_messages: false
  capabilities:
    vertical_swing: true
    horizontal_swing: true
    presets: 
      quiet: true
      windfree: true
      fast: true
  devices:
    - address: "20.00.00"
      climate:
        name: None
      room_temperature_offset: -2.0
      automatic_cleaning:
        name: "Automatic cleaning"
    - address: "10.00.00"
      outdoor_temperature:
        name: "Outdoor temperature"
      outdoor_instantaneous_power:
        name: "Power"
        accuracy_decimals: 0
      outdoor_cumulative_energy:
        name: "Energy"
        accuracy_decimals: 2
```

omerfaruk-aran, 2024-11-06:
Hello, I could not observe the problem you mentioned in my NASA air conditioning system. Log records will help better. I'm waiting for your log records. Thanks.

ouchkilljoys, 2024-11-07:
I've had this same issue since I've setup mine, its like it recieves an "empty" message from the unit which overrides what the ESPhome has sent

mbo18, 2024-11-08:
Ok so there is no issue 🫢
It works fine, it is just that it takes 13 seconds to power my HVAC: a first packet is sent that do not power on the HVAC then 13 seconds later another packet is sent and the HVAC goes on. 
I was not waiting enough… On HA HVAC would go from Off to heat then off again (that’s why I was thinking that it was not working) and 13s later to heat. 
Closing this issue. Thank you

closed 2024-11-08 by mbo18