omerfaruk-aran/esphome_samsung_hvac_bus #230: Virus Doctor function

jjprz, 2024-12-01:
### The problem

My unit has the ability to run Virus Doctor. Could this option be implemented? I can help with anything.

[image]

### ESPHome Version

2024.10.0

### Samsung HVAC Component Version

AC052RNMDKG

### Home Assistant Version

2024.11.0

### AC Type

NASA

### Indoor AC Unit Model

AC052RNMDKG

### Outdoor AC Unit Model

AC052RXADKG

### ESP Device Model

M5STACK ATOM Lite + M5STACK RS-485

### Connection Points

F1/F2

omerfaruk-aran, 2024-12-28:
Hello @jjprz,

Thank you for bringing this up! Unfortunately, we do not have access to a device with the "Virus Doctor" feature, which limits our ability to test or log the functionality directly.

If you could share the relevant logs when attempting to activate or use this feature, we would be happy to assist in integrating it into the system. Any additional details about how the function behaves or interacts with the controller would also be very helpful.

Looking forward to your input!

Best regards,

jjprz, 2026-10-31:
I finally captured the bus on my unit (NASA, AC052RNMDKG / AC052RXADKG). Virus Doctor is the SPI ionizer:

- `0x4023` (`NASA_USE_SPI`): whether the unit has it. My unit always reports `0`.
- `0x4043` (`NASA_CONTROL_SPI`): on/off. When I send `0x4043 = 1` from the ESP, the indoor unit immediately answers `0x4043 = 0`, so it rejects it.

So my model doesn't have the ionizer. The wired controller shows the option anyway (its manual says the icon blinks if the unit doesn't support it). I can't test it further, so feel free to close this, or keep it open for someone with an SPI-capable unit: reading `0x4023` and writing `0x4043` should be enough.

closed 2026-11-01 by jjprz