omerfaruk-aran/esphome_samsung_hvac_bus #242: `Bug` [Samsung_HVAC]: Does not compile with ESPHome 2024.12.x And esp-idf framework

mbo18, 2024-12-28:
### The problem

I’m unable to compile the code since ESPHome 2024.12.x. It was compiling correctly with ESPHome 2024.11.x

After some tests, it does compile when using "arduino" as framework type but no more with "esp-idf"

See attached logs 

### ESPHome Version

2024.12.2

### Samsung HVAC Component Version

Latest

### Home Assistant Version

2024.12.5

### AC Type

NASA

### Indoor AC Unit Model

AR09TXCAAWKN

### Outdoor AC Unit Model

AR09TXCAAWKX

### ESP Device Model

M5stack atom lite + m5stack rs485

### Connection Points

F1/F2

### ESPHome Logs

[logs_salon-hvac_run.txt](https://github.com/user-attachments/files/18265728/logs_salon-hvac_run.txt)

### YAML Configuration

```yaml
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@main
    components: [samsung_ac]

samsung_ac:
  debug_log_messages_raw: false
  debug_log_messages: false
  capabilities:
    vertical_swing: true
    horizontal_swing: true
    presets: 
      quiet: true
      windfree: true
      fast: true
  devices:
    - address: "20.00.00"
      climate:
        name: None
      room_temperature_offset: -2.0
      automatic_cleaning:
        name: "Automatic cleaning"
    - address: "10.00.00"
      outdoor_temperature:
        name: "Outdoor temperature"
      outdoor_instantaneous_power:
        name: "Power"
        accuracy_decimals: 0
      outdoor_cumulative_energy:
        name: "Energy"
        accuracy_decimals: 2
```

omerfaruk-aran, 2024-12-28:
Hi @mbo18 ,

I’ve created a PR (#243) addressing the issue you reported regarding the compilation failure with ESPHome 2024.12.x when using the ESP-IDF framework. The changes involve updating the `debug_mqtt.cpp` file to align with the new ESP-IDF API while maintaining backward compatibility with the Arduino framework.

I’d like you to test with your setup to confirm it resolves the issue.

```yaml
external_components:
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@branch-name
    components: [samsung_ac]

samsung_ac:

  # Sends all NASA package values to MQTT so the can be analysed or monitored.
  debug_mqtt_host: 192.168.1.100
  debug_mqtt_port: 1883
  debug_mqtt_username: mqtt_user
  debug_mqtt_password: mqtt_pass

    ...

```

closed 2024-12-28 by omerfaruk-aran

mbo18, 2024-12-28:
Hi @omerfaruk-aran,

Thank you very much for your work and reply. It works now! 👍

PR #243 (2024-12-28): Fix: Update MQTT client for ESP-IDF compatibility
## 📄 Description
### What does this Pull Request do?

- [x] 🐞 Bug Fix
- [ ] ✨ New Feature
- [ ] 🔨 Code Improvement
- [ ] 📚 Documentation Update

This PR resolves the compatibility issues with the `debug_mqtt.cpp` file when using ESP-IDF framework in ESPHome 2024.12.x. The changes include updates to the `esp_mqtt_client_config_t` structure to align with the updated ESP-IDF framework API.

### ✨ Key Changes

1. Replaced deprecated `mqtt_cfg.host`, `mqtt_cfg.port`, `mqtt_cfg.username`, and `mqtt_cfg.password` members with the updated configuration [... cut]