omerfaruk-aran/esphome_samsung_hvac_bus #256: Lighting option on Windfree models

llubi23, 2025-01-20:
### The problem

Some of the new EU models, have a 8 segment to show actual temperature and desired one. In night use it's too bright.

It can be turned off using the remote or the SmartThings app, under the word "Lighting"

Tried to put on my YAML file but won't work, can it be added to new versions?

### ESPHome Version

2024.12.4

### Samsung HVAC Component Version

20240213.1

### Home Assistant Version

2025.1.2

### AC Type

NASA

### Indoor AC Unit Model

AR12TXFCAWKN

### Outdoor AC Unit Model

AJ050TXJ2KG/EU

### ESP Device Model

M5STACK ATOM LITE +  M5STACK RS-485

### Connection Points

F1/F2

### YAML Configuration

```yaml

```

TimPasquini, 2025-01-21:
This is the last forcing a lot of us to keep smartthings connected. Unfortunately, lighting and sound seems to be handled in the head-unit directly and doesn't appear to get published to the F1/F2 bus.

#144 has on-going exploration of controlling the audible beep when you make selections. Both the beeps from the head unit and the lights are under the same menu on the smartthings app. Also discussion #164.

The head-units are what get connect to wifi, so anything coming through the smartthings app and api gets delivered directly to the head-unit. The lights and beeping are settings for the head-units themselves so when they receive the message, they implement them directly and don't publish anything to the F1/F2 bus. The outdoor unit or other head units don't have any reason to know the state of the display light or the beeps so there is no reason to publish it.

llubi23, 2025-01-22:
OK, let's hope they update the code and we can modify it :)

In the meantime, I remember to read about a configuration of ECO mode in the outside unit than turn off the light. I'll check.

Thanks!

omerfaruk-aran, 2025-03-09:
Hello,

While performing this operation on the SmartThings side, could you please check the logs on the ESP? Could you observe if any packets are being sent/received? I would appreciate it if you could share the log file as well for further investigation.

karelvhouten, 2025-07-20:
Hi,
I just tested light / sound control with my NASA devices (control from app, logging of unknown modbus packets), and got this trace:
During this trace, I switched the LED and sound off, and on again.

```
[13:47:55][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 4}
[13:47:55][W][samsung_ac:802]: Undefined s:20.00.02 d:b0.00.ff Variable 4202 = 190
[13:47:55][W][samsung_ac:590]: s:20.00.02 d:b0.00.ff VAR_in_temp_room_f 23.3
[13:47:55][W][samsung_ac:705]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_in_f 22.2
[13:47:55][W][samsung_ac:712]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_out_f 21
[13:48:05][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 15}
[13:48:05][W][samsung_ac:802]: Undefined s:20.00.02 d:b0.00.ff Variable 4202 = 190
[13:48:05][W][samsung_ac:590]: s:20.00.02 d:b0.00.ff VAR_in_temp_room_f 23.3
[13:48:06][W][samsung_ac:705]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_in_f 22.2
[13:48:06][W][samsung_ac:712]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_out_f 21
[13:48:08][D][device_state_tracker:046]: No change in value for device: 20.00.02
[13:48:08][C][samsung_ac:057]: Configured devices: 10.00.00, 20.00.00, 20.00.01, 20.00.02
[13:48:08][C][samsung_ac:071]:   Indoor:  20.00.00, 20.00.01, 20.00.02
[13:48:15][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b3.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 26}
[13:48:15][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Enum 23 = 0
[13:48:15][W][samsung_ac:622]: s:20.00.02 d:b3.00.ff ENUM_in_operation_power 0
[13:48:15][W][samsung_ac:640]: s:20.00.02 d:b3.00.ff ENUM_in_operation_mode 1
[13:48:15][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Enum 4002 = 1
[13:48:15][W][samsung_ac:652]: s:20.00.02 d:b3.00.ff ENUM_in_fan_mode 0
[13:48:15][W][samsung_ac:617]: s:20.00.02 d:b3.00.ff ENUM_in_state_humidity_percent 67
[13:48:15][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Enum 4132 = 3
[13:48:15][W][samsung_ac:597]: s:20.00.02 d:b3.00.ff VAR_in_temp_target_f 19
[13:48:15][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 4202 = 190
[13:48:15][W][samsung_ac:590]: s:20.00.02 d:b3.00.ff VAR_in_temp_room_f 23.3
[13:48:16][W][samsung_ac:705]: s:20.00.02 d:b3.00.ff VAR_in_temp_eva_in_f 22.2
[13:48:16][W][samsung_ac:712]: s:20.00.02 d:b3.00.ff VAR_in_temp_eva_out_f 21
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 421c = 0
[13:48:16][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b3.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 27}
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 4285 = 0
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 429d = 0
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 42d1 = 65535
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 42d2 = 65535
[13:48:16][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Variable 42d3 = 65535
[13:48:22][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b3.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 35}
[13:48:22][W][samsung_ac:802]: Undefined s:20.00.02 d:b3.00.ff Structure #4623 = 4
[13:48:25][W][samsung_ac:825]: MSG: #Packet Src:20.00.02 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 39}
[13:48:25][W][samsung_ac:802]: Undefined s:20.00.02 d:b0.00.ff Variable 4202 = 190
[13:48:25][W][samsung_ac:590]: s:20.00.02 d:b0.00.ff VAR_in_temp_room_f 23.3
[13:48:25][W][samsung_ac:705]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_in_f 22.2
[13:48:25][W][samsung_ac:712]: s:20.00.02 d:b0.00.ff VAR_in_temp_eva_out_f 21
```

Hope this helps.
Regards,
Karel.

zeratul021, 2025-08-13:
If it helps, this is my log from model AR12TXEAAWKN/EU where I turned Lighting off and then on via SmartThings.

[logs_esphome-samsung-hvac-bus_run.txt](https://github.com/user-attachments/files/21745571/logs_esphome-samsung-hvac-bus_run.txt)

kp-bit, 2026-08-31:
> Any update on this? The remote and Tasmota IR are able to send the IR command to turn off the display. Anything I can do to help? The display is like a bright night light and definitely needs to be turned off 😅

Does the Tasmota IR change only the display light or all kinds of other stuff in the same keypress? Are you able to export the Pronto codes from the Tasmota?

I would very much like to try the Pronto codes in the HAIR integration where I can capture the IR commands from the Samsung remote, but they are pretty unusable because it bundles commands.

zeratul021, 2026-09-22:
> in my windfree systems the beep off and volume 0 are not stored after the power off (not standby)

you mean like turning off power to the outside unit? When I said between restarts I meant it either setting Climate to off via HA or pressing the red button on the IR remote.

zeratul021, 2026-11-02:
@omerfaruk-aran any chance you will find some time to pick this up and at leat have look if there is something interesting visible in the logs?

roquerodrigo, 2026-11-02:
After months of trying to get this working, today I finally managed to control the panel display over F1/F2. I implemented it in my private fork, and it is working.

The display is controlled by NASA message **`0x4016`** (Enum): `0` = off, `1` = on. It is not in the message table of this project, and passive logging cannot find it because the indoor unit never sends it in a Notification. It only appeared when I actively read the `0x4000`–`0x41FF` range with the panel on and off and compared the responses.

**Reading**

Send `Read` (DataType 1) to the indoor unit. It answers with `Response` (DataType 5). In the same sweep, unsupported Enum messages returned `0xFF`.

```
32001162ff00200002c0110101401600378b34    read 0x4016 from 20.00.02
```

**Writing**

Send `Request` (DataType 3) with the value. The unit answers with `Ack`, and the panel changes with one beep. `Write` (DataType 2) also works and beeps as well. Both `62.ff.00` and `80.ff.00` were accepted as source addresses.

```
32001162ff00200002c0130201401600521934    write 0x4016 = 0 to 20.00.02
```

The packet number and CRC change from one frame to the next.

**Keeping the panel off when the unit powers on**

Powering on resets the value to `1`, which is why the panel keeps coming back. Setpoint and fan changes left the value at `0`, and a mode change left the panel off.

The unit only keeps the panel off if the power-on packet actually changes the value:

1. With the unit off, write `0x4016 = 1`.
2. Power on with a single `Request` that has `0x4016 = 0` **ahead of** `0x4000 = 1`, for example `4016=0, 4001=mode, 4000=1`.

The panel then stays off from the start, with only the normal power-on beep. The same packet with `0` already stored did not work: the value went back to `1` and the panel came on.

When the unit is powered on by the remote, the component only finds out afterwards, so the panel lights up and has to be turned off with a normal write, which costs one extra beep.

**What this was tested on**

One outdoor unit with three indoor units on F1/F2, without built-in Wi-Fi. I do not know the exact indoor model. The display changed on two of the units with someone watching the panel. The third one answers the read, but I have not confirmed it visually. It would be useful to hear from people with other models.

**Next steps**

You can replicate this in the original project from the details above, or let me know if you want my code. I would need to work on it first to remove what is hardcoded for my private setup.

TimPasquini, 2026-11-02:
> It is not in the message table of this project, and passive logging cannot find it because the indoor unit never sends it in a Notification

I've long suspected this was the case. Glad to see there is some sort of bridge between NASA and the SPI bus. Nice work finding it!

I will give your fork a try and see if I can get it working!

AnilEkmekci, 2026-11-02:
@roquerodrigo I make changes on my own fork with ai using your findings an it is working.

zeratul021, 2026-11-03:
@roquerodrigo I will try to this next weekend. Great investigation on your end! 
Did you find the same for turning off sound beeps (those are persisted so less needed).

omerfaruk-aran, 2026-11-04:
Thanks so much to @roquerodrigo for tracking down `0x4016` and sharing the packet details, and to @zeratul021, @karelvhouten, and everyone else who helped with logs and testing. It really helped move this forward.

I've now added optional NASA display-light control in #405. I tested it on my own indoor unit: I can turn the panel display off and on from Home Assistant without affecting the AC, and the standard toggle is working too. As noted here, the display turns back on when the AC is powered on; keeping it off automatically is not part of this PR yet.

The PR is ready for review, but hasn't been merged. I'll leave this issue open until we've reviewed the change and can confirm the next steps. Thanks again for everyone's patience and help!

okin-88, 2026-11-04:
Great find @roquerodrigo and thanks to @omerfaruk-aran for the quick implementation!

Would it be possible to add an optional refresh button directly under `display_light`, similar to `filter_reset`, like in the following example?

```yaml
display_light:
  name: "Kitchen display light"
  state_valid:
    name: "Kitchen display light state valid"
  refresh:
    name: "Kitchen refresh display light"
```

This would allow scripts in ioBroker or HA to manually refresh the display state after it has been changed using the remote control or SmartThings app.

omerfaruk-aran, 2026-11-06:
Hi @okin-88, thanks for the suggestion!

The optional **display-light refresh button** has now been implemented and merged into `main` via #407. I've also tested it successfully on my own setup.

You can enable it under the existing `display_light` configuration:

```yaml
display_light:
  name: "Kitchen display light"
  state_valid:
    name: "Kitchen display light state valid"
  refresh:
    name: "Kitchen refresh display light"
```

Pressing the button sends a single NASA `Read 0x4016` request and updates the Home Assistant switch state from the verified response. This is useful after changing the lighting through the IR remote or SmartThings.

The refresh action **only reads the current state**: it does not turn the display on or off, and it does not introduce periodic polling. Existing configurations without `refresh` continue to work as before.

Thanks again for the feedback and for helping improve the integration!

okin-88, 2026-11-06:
Great, it works exactly as expected - thanks for the addition! :raised_hands:
FYI: the option isn't listed in `example.yaml` yet...

(I can't say if it was my fault, but the main branch didn't work for me, but `source: github://omerfaruk-aran/esphome_samsung_hvac_bus@pull/407/head` worked fine)

omerfaruk-aran, 2026-11-06:
Hi @okin-88, thanks for catching this — you're right! The optional `display_light.refresh` setting was missing from the repository's `example.yaml`.

I've added the commented configuration example and a short explanation of what the button does. The documentation update was merged in #408, so `example.yaml` on `main` now includes:

```yaml
# display_light:
#   name: "Kitchen display light"
#   state_valid:
#     name: "Kitchen display light state valid"
#   refresh:
#     name: "Kitchen refresh display light"
```

The button performs a one-shot read of `0x4016`, without changing the panel or adding periodic polling.

Regarding the `@main` issue: I verified that the refresh implementation and YAML schema are present on `main`. ESPHome caches Git-based external components (the default refresh interval is one day), which **might** explain why `@pull/407/head` worked while your local `@main` source did not. To force a fresh check while testing, you can temporarily use:

```yaml
external_components:
  - source: github://omerfaruk-aran/esphome_samsung_hvac_bus@main
    refresh: 0s
    components: [samsung_ac]
```

If `@main` still fails after refreshing, please share the exact ESPHome error output so we can investigate it separately.

Thanks again for testing the feature and reporting the documentation gap!

TimPasquini, 2026-11-06:
Just wanted to report I have working display control for my windfree models. 

AJ036BXS4CH with four attached AR07TSFABWKN head units

AR09TSFACWKX with one attached AR09TSFACWKN

karelvhouten, 2026-11-07:
I can also confirm the display control working on my 3 windfree indoor units, controlled from Openhab via Mosquitto to the ESPhome module on the RS-485 bus.

Lots of thanks to all your efforts to add this to this excellent piece of software!