omerfaruk-aran/esphome_samsung_hvac_bus #266: `Feature` [Samsung_HVAC]: RS485 driver enable

Niaxor, 2025-03-01:
### The problem

I'd like to be able to use a regular RS485 transceiver IC with a ESP-12F with this project, as a cheaper alternative to the recommended units.

The one thing that could make this more feasible is the option for a driver enable GPIO output for when the Samsung esphome HVAC component is writing to UART, as most transceiver ICs require this, without additional circuit complexity for automatic driver enable. 

Included below is an example RS485 transceiver that also happens to be found in my Samsung wired remote for my HVAC unit

https://www.alldatasheet.com/datasheet-pdf/download/931797/TI1/VP3082.html

### ESPHome Version

Any

### Samsung HVAC Component Version

Any

### Home Assistant Version

Any

### AC Type

NON NASA

### Indoor AC Unit Model

Any

### Outdoor AC Unit Model

Any

### ESP Device Model

ESP-12F, VP208

### Connection Points

F1/F2

### YAML Configuration

```yaml

```

haileys, 2025-04-03:
This is possible with the esp-idf UART driver's built-in RS485 half-duplex mode, and then configuring the RTS pin to the pin that's connected to the RS485 transceiver's drive enable pin.

This esphome yaml does it:

```yaml
  on_boot:
    - priority: 999 # BUS - 1.0
      then:
        - lambda: |-
            uart_port_t uart = (uart_port_t)(id(samsung_uart).get_hw_serial_number());

            if (esp_err_t err = uart_set_pin(uart, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, GPIO_NUM_19, UART_PIN_NO_CHANGE)) {
              ESP_LOGE("samsung_ac_rs485", "uart_set_pin for RTS failed: %s", esp_err_to_name(err));
            }

            if (esp_err_t err = uart_set_mode(uart, UART_MODE_RS485_HALF_DUPLEX)) {
              ESP_LOGE("samsung_ac_rs485", "uart_set_mode failed: %s", esp_err_to_name(err));
            }
```

Ideally this functionality could be exposed upstream in esphome's uart component.

Niaxor, 2025-06-03:
> (omerfaruk-aran/esphome_samsung_hvac_bus#266)
>
> Is #281
> what are you looking for?

It absolutely is! Thanks so much.

Looking forward to testing this soon 🙂