omerfaruk-aran/esphome_samsung_hvac_bus #270: problem only Tx

wifi75, 2025-04-04:
### The problem

hello I have an external unit and 2 internal ones, 

the problem is that I can view all the changes that occur via the remote control via HA, but if I try to turn on or off or change the temperature from HA, these do not work. it seems that the tx does not work.

I currently only have one external drive enabled.

What can I check?

### ESPHome Version

2025.3.3 

### Samsung HVAC Component Version

20240213.1

### Home Assistant Version

2025.4.0

### AC Type

NON NASA

### Indoor AC Unit Model

AR12HSSDBWKNEU

### Outdoor AC Unit Model

AJ040FCJ2EH

### ESP Device Model

M5STACK ATOM Lite + M5STACK ATOM RS-485 Base

### Connection Points

f1,f2

### ESPHome Logs

[logs_samsung_logs.txt](https://github.com/user-attachments/files/19601527/logs_samsung_logs.txt)

### YAML Configuration

```yaml
esphome:
  name: ac-rooms
  friendly_name: ac-rooms

esp32:
  board: m5stack-atom
  

web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR # Remove the "Your component takes to long to respond warning"

api:

ota:
  - platform: esphome

# Setup your WiFi password
wifi:
  ssid: "WiFi" # !!!CHANGE ME!!!
  password: "12345678901234567890" # !!!CHANGE ME!!!
  manual_ip:
    static_ip: 192.168.1.168
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

captive_portal:

#uart:
#  tx_pin: GPIO19 # Use GPIO26 for ATOM Tail485 
#  rx_pin: GPIO22  # Use GPIO32 for ATOM Tail485
#  baud_rate: 9600
#  parity: EVEN

#################################
#### UART SETUP/DECODE BYTES ####
#################################

uart:
  id: uart_bus
  tx_pin: 19
  rx_pin: 22
  baud_rate: 9600
  data_bits: 8
  stop_bits: 1
  parity: EVEN
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ' ');

external_components:
  - source: github://tristone-cz/esphome_samsung_hvac_bus@patch-slow-registration  
    components: [samsung_ac]
    refresh: 0s

# Configuration of AC component

samsung_ac:
  non_nasa_keepalive: false
  debug_log_undefined_messages: true
  debug_log_messages: true
  capabilities: 
    vertical_swing: true
    presets: 
      quiet: true
      fast: true
      windfree: true

  devices:
    - address: "c8" # Outdoor device address

    - address: "00" # Indoor device address 
      capabilities:
        horizontal_swing: true  
      climate:
        name: "Soggiorno AC"
      room_temperature:
        name: "Soggiorno AC temperature"
      target_temperature:
        name: "Soggiorno AC target"
      power:
        name: "Soggiorno AC power"
      mode:
        name: "Soggiorno AC mode"
      room_temperature_offset: 0
      error_code:
        name: "Error Code"
 #   - address: "01" # Indoor device address 
 #     capabilities:
 #       horizontal_swing: true  
 #     climate:
 #       name: "Matrimoniale AC"
 #     room_temperature:
 #       name: "Matrimoniale AC temperature"
 #     target_temperature:
 #       name: "Matrimoniale AC target"
 #     power:
 #       name: "Matrimoniale AC power"
 #     mode:
 #       name: "Matrimoniale AC mode"
 #     room_temperature_offset: 0
 #     error_code:
 #       name: "Error Code"
```

brazoayeye, 2025-04-04:
It seems an hardware failure, check if TX pin (19) is well welded.

Try to remove the RS485 adapter and set TX/RX pin high/low and test them with a multimeter to understand if they works well
https://esphome.io/components/switch/gpio.html

brazoayeye, 2025-04-04:
Ciao

I think something between the cpu and the reciver in this line

[image]

[image]

AB is half duplex and differential, so it doesn't make sense that something there could cause your issue.
The pin 19 transfer informations from the cpu to the rs485 chip, while the 22 from the rs458 to the cpu so I suppose the problem is in the 19 one or in the rs485 chip itself.

You can understand which component is broken (if it is - maybe it's something else) uploading a simple software with a swich, inseriting a small wire in the position pointed by the arrow in the second image, and testing between it and GND if you find 3.2V with swith on and 0V when off.

Or maybe if you have different hardware you can simply test changing one and the other.

wifi75, 2025-04-04:
thank you so much for your precious advice, i will do some tests tomorrow morning

wifi75, 2025-04-05:
> Ciao
> 
> Penso che ci sia qualcosa tra la CPU e il ricevitore in questa linea
> 
> [image]
> 
> [image]
> 
> AB è half duplex e differenziale, quindi non ha senso che qualcosa lì possa causare il tuo problema. Il pin 19 trasferisce informazioni dalla CPU al chip rs485, mentre il 22 dall'rs458 alla CPU, quindi suppongo che il problema sia nel pin 19 o nel chip rs485 stesso.
> 
> Puoi capire quale componente è rotto (se lo è, forse è qualcos'altro) caricando un semplice software con un interruttore, inserendo un piccolo filo nella posizione indicata dalla freccia nella seconda immagine e testando tra questo e GND se trovi 3,2 V con l'interruttore acceso e 0 V quando è spento.
> 
> Oppure, se hai hardware diverso, puoi semplicemente provare a cambiare l'uno e l'altro.

hello i did a test trying to upload a code on the atom lite cpu, and both pins 23 and 19 work great, when the switch is off or 0, while if the switch is on i have the voltage of 3.2.v. i tested both pins!
so i think the defect is on the RS.485 module correct?

wifi75, 2025-04-15:
> Ciao
>
> I think something between the cpu and the reciver in this line
>
> [image]
>
> [image]
>
> AB is half duplex and differential, so it doesn't make sense that something there could cause your issue. The [...]

Hi I replaced the rs485 module, unfortunately it still doesn't work.. I can't send commands to the climate.
at this point could it be a code problem?

brazoayeye, 2025-04-15:
Do you see errors in the console when you change values from HASS?
Errors are usually red and always printed so I suggest to set debug_log_undefined_messages: false to avoid too many rows.

wifi75, 2025-04-16:
@brazoayeye Hi, what can I try to do?

brazoayeye, 2025-04-16:
I really don't know, I have less packets (1/3s) while you have more than 10 messages/s but i have a NASA device.

You could try to disconnect few internal units to see if something changes, but I don't have many hopes because if the problem was collisions writing would work sometimes...

brazoayeye, 2025-04-16:
I don't think, but give a try. 

Have you tested the other repo ? https://github.com/omerfaruk-aran/esphome_samsung_hvac_bus/issues/102#issuecomment-2774914866 

Or maybe test an older one like mine : github://brazoayeye/esphome_samsung_ac@main

wifi75, 2025-04-16:
Maybe I should abandon the idea.. and yet I'm one step away from making it work... who knows what it will be.