omerfaruk-aran/esphome_samsung_hvac_bus #271: `Bug` [Samsung_HVAC]: Openhab via MQTT: preset-mode receives command (not string) 'none'.

karelvhouten, 2025-04-09:
### The problem

Samsung NASA device connected via ESPHome and MQTT to Openhab. When indoor unit is off, the <device>#preset-mode mqtt channel keeps receiving a 'none' value, which is not a valid type for a text string. As soon as the indoor unit is on, and a specific preset is selected, a string is send, for example "quiet". Then all is well. How can I translate the 'no preset selected' situation to the STRING "None" (not an empty value).

Openhab gives this warning message about every second:
`2025-04-09 16:22:50.687 [WARN ] [ab.binding.mqtt.generic.ChannelState] - Command 'none' from channel 'mqtt:homeassistant:mosquitto:samsung-ac:slaapkamer_ac#preset-mode' not supported by type 'TextValue': Value none not within range`

Thanks in advance!

### ESPHome Version

2025.2.2q

### Samsung HVAC Component Version

Current main

### Home Assistant Version

N.a. Openhab version

### AC Type

NASA

### Indoor AC Unit Model

AR09TXFCAWKN

### Outdoor AC Unit Model

AJ068TXJ3KG-EU

### ESP Device Model

M5STACK ATOM Lite + M5STACK RS485

### Connection Points

F1/F2 in outside unit

### Samsung AC Logs

[16:28:58][C][samsung_ac:053]: Configured devices: 10.00.00, 20.00.00, 20.00.01, 20.00.02
[16:28:58][C][samsung_ac:065]: Discovered devices:
[16:28:58][C][samsung_ac:066]:   Outdoor: 10.00.00
[16:28:58][C][samsung_ac:067]:   Indoor:  20.00.00, 20.00.01, 20.00.02
[16:28:59][W][samsung_ac:594]: s:20.00.00 d:b0.00.ff VAR_in_temp_room_f 18.8
[16:28:59][D][sensor:094]: 'Caroline AC temperature': Sending state 18.80000 °C with 1 decimals of accuracy
[16:28:59][D][climate:396]: 'Caroline AC' - Sending state:
[16:28:59][D][climate:399]:   Mode: OFF
[16:28:59][D][climate:404]:   Fan Mode: AUTO
[16:28:59][D][climate:410]:   Preset: NONE
[16:28:59][D][climate:416]:   Swing Mode: OFF
[16:28:59][D][climate:419]:   Current Temperature: 18.80°C
[16:28:59][D][climate:425]:   Target Temperature: 21.00°C
[16:28:59][W][samsung_ac:709]: s:20.00.00 d:b0.00.ff VAR_in_temp_eva_in_f 19.4
[16:28:59][W][samsung_ac:716]: s:20.00.00 d:b0.00.ff VAR_in_temp_eva_out_f 19.3
[16:28:59][W][samsung_ac:725]: s:10.00.00 d:b0.00.ff VAR_out_error_code 0
[16:28:59][D][sensor:094]: 'error_code': Sending state 0.00000  with 0 decimals of accuracy
[16:28:59][W][samsung_ac:594]: s:20.00.01 d:b0.00.ff VAR_in_temp_room_f 22.8
[16:28:59][D][sensor:094]: 'Logeerkamer AC temperature': Sending state 22.80000 °C with 1 decimals of accuracy
[16:28:59][D][climate:396]: 'Logeerkamer AC' - Sending state:
[16:28:59][D][climate:399]:   Mode: OFF
[16:28:59][D][climate:404]:   Fan Mode: AUTO
[16:28:59][D][climate:410]:   Preset: NONE
[16:28:59][D][climate:416]:   Swing Mode: OFF
[16:28:59][D][climate:419]:   Current Temperature: 22.80°C
[16:28:59][D][climate:425]:   Target Temperature: 24.00°C
[16:28:59][W][samsung_ac:709]: s:20.00.01 d:b0.00.ff VAR_in_temp_eva_in_f 23
[... 172 lines cut ...]
      # The captured value represents the current power draw of the outdoor HVAC components, helping track energy usage patterns.
      outdoor_instantaneous_power:
        name: "Outdoor Instantaneous Power"

      # This sensor records the cumulative energy consumption of the outdoor unit in kWh.
      # It calculates the total energy consumed over time, allowing users to monitor and analyze energy efficiency.
      outdoor_cumulative_energy:
        name: "Outdoor Cumulative Energy"

      # This sensor measures the current drawn by the outdoor unit in Amperes.
      # Monitoring current values helps identify electrical irregularities and ensure safe power levels in the system.
      outdoor_current:
        name: "Outdoor Current"

      # This sensor tracks the voltage supplied to the outdoor unit in Volts.
      # Consistent voltage readings indicate stable power delivery, while deviations can help detect electrical issues in the system.
      outdoor_voltage:
        name: "Outdoor Voltage"
```

### Additional Details

Otherwise the integration works perfect, I can read and control my units from Openhab just fine. Only the log file is flooded with these warnings.

Thanks for this very helpfull software!

karelvhouten, 2025-06-03:
Hi, thanks for your reply!

Further troubleshooting pointed indeed to a misinterpretation by the Openhab MQTT binding. As the error is from Openhab, not ESPHome, I'm afraid solution 2 is not an option, I could try to suppress the error via log4j in Openhab, but would not solve the issue that I can't reset the "preset" to "empty", because Openhab will not send that as a string. 

What I was looking for, is a way to adapt the ESPHome module in a way that instead of "none" the string "no-preset" (or something like that) could be used (as a status message from ESPHome to Openhab, AND as a command from openhab to ESPHome.

If you could point me the source module where this translation could be done, I can do some experiments myself.

Regards,
Karel.

omerfaruk-aran, 2025-06-03:
Hi Karel,

Thanks a lot for your follow-up and clear explanation — really helpful!

You're absolutely right: the issue is with OpenHAB's MQTT `TextValue` binding not accepting `"none"` as a valid option. Replacing `"none"` with something like `"no-preset"` is technically possible, but it would deviate from the ESPHome climate spec, which uses `"none"` internally to represent "no active preset".

That said, if you’re fine with maintaining a custom build for OpenHAB, I can point you to where the value is set:

### 🔧 File:
`components/samsung_ac/samsung_ac_climate.cpp`

### 📍 Line to modify:
Look for this part in the `control_to_climate` or `publish_state` method:
```cpp
this->preset = climate::CLIMATE_PRESET_NONE;
```

You can change it to:
```cpp
this->preset = "no-preset";
```

But note that:

- Home Assistant may not understand "no-preset" unless it's patched too.
- You should test bi-directional preset handling (commands from OpenHAB as well).

If there's enough community interest, we could even consider a YAML config flag to override the default "none" string for broader OpenHAB support.

Let me know if you’d like help patching or testing your fork. Happy to support your experiments!

Best regards,

karelvhouten, 2025-07-20:
The problem turned out to be the way Openhab interpreted the auto-discovery info from ESPhome. When I configure the channels manually, there are no errors and everything works fine.

Case closed. ;-)

closed 2025-07-20 by karelvhouten