omerfaruk-aran/esphome_samsung_hvac_bus #287: `Bug` [Samsung_HVAC]: Can't set temp in Cool below 18c

BlairC1, 2025-06-19:
### The problem

Hi, first of all, great project, super happy with it. I'm having an issue with setting cool mode lower than 18, the AC will beep but remain at its previous temperature setting. Interestingly if I set it to heat mode it works correctly, if I then switch back to cool it'll remain at that temperature eg. 17c. Is this a known issue?

Btw. not sure if this unit is Nasa or not, it lists as nasa on the check tool but ids are only 2 digits.

### ESPHome Version

2025.6.0

### Samsung HVAC Component Version

20240213.1

### Home Assistant Version

2025.6.1

### AC Type

NASA

### Indoor AC Unit Model

AR09FSSSCWKNSA

### Outdoor AC Unit Model

AR09FSSSCWKX

### ESP Device Model

M5stack Atom Lite

### Connection Points

F1/F2

### ESPHome Logs

[16:48:52][D][climate:396]: 'Bedroom AC' - Sending state:
[16:48:52][D][climate:399]:   Mode: COOL
[16:48:52][D][climate:404]:   Fan Mode: AUTO
[16:48:52][D][climate:410]:   Preset: NONE
[16:48:52][D][climate:416]:   Swing Mode: OFF
[16:48:52][D][climate:419]:   Current Temperature: 21.00°C
[16:48:52][D][climate:425]:   Target Temperature: 20.00°C
[16:48:52][D][samsung_ac:512]: Sending controller registration request...
[16:48:52][W][samsung_ac:097]: write 32d0c8d1d2000000000000001b34
[16:48:53][D][samsung_ac:512]: Sending controller registration request...
[16:48:53][W][samsung_ac:097]: write 32d0c8d1d2000000000000001b34
[16:48:53][D][climate:011]: 'Bedroom AC' - Setting
[16:48:53][D][climate:040]:   Target Temperature: 16.00
[16:48:53][W][samsung_ac:097]: write 32d000b01f151001f4210000ae34
[16:48:54][W][samsung_ac:027]: update
[16:48:54][D][device_state_tracker:046]: No change in value for device: 00
[16:48:54][C][samsung_ac:057]: Configured devices: 00, c8
[16:48:54][C][samsung_ac:069]: Discovered devices:
[16:48:54][C][samsung_ac:070]:   Outdoor: c8
[16:48:54][C][samsung_ac:071]:   Indoor:  00
[16:48:54][D][samsung_ac:512]: Sending controller registration request...
[16:48:54][W][samsung_ac:097]: write 32d0c8d1d2000000000000001b34
[16:48:54][D][samsung_ac:512]: Sending controller registration request...
[16:48:54][W][samsung_ac:097]: write 32d0c8d1d2000000000000001b34
[16:48:55][D][number:012]: 'Bedroom AC target temperature': Sending state 20.000000
[16:48:55][D][climate:396]: 'Bedroom AC' - Sending state:
[16:48:55][D][climate:399]:   Mode: COOL
[16:48:55][D][climate:404]:   Fan Mode: AUTO
[16:48:55][D][climate:410]:   Preset: NONE
[16:48:55][D][climate:416]:   Swing Mode: OFF
[16:48:55][D][climate:419]:   Current Temperature: 21.00°C
[16:48:55][D][climate:425]:   Target Temperature: 20.00°C
[16:48:55][D][sensor:099]: 'Bedroom AC temperature': Sending state 21.00000 °C with 1 decimals of accuracy
[16:48:55][D][climate:396]: 'Bedroom AC' - Sending state:
[16:48:55][D][climate:399]:   Mode: COOL
[16:48:55][D][climate:404]:   Fan Mode: AUTO
[16:48:55][D][climate:410]:   Preset: NONE
[16:48:55][D][climate:416]:   Swing Mode: OFF
[16:48:55][D][climate:419]:   Current Temperature: 21.00°C
[16:48:55][D][climate:425]:   Target Temperature: 20.00°C

### YAML Configuration

```yaml

```

omerfaruk-aran, 2025-07-16:
Thanks for the detailed report — very helpful!

From the logs it looks like ESPHome is correctly sending the command to set 16°C in cool mode (`write 32d000b01f151001f4210000ae34`). The AC is receiving it (you hear the beep) but it doesn't accept the temperature change and stays at the previous setting.

This suggests that your particular unit likely has a firmware limitation: some Samsung models enforce a minimum temperature limit of 18°C in cool mode, regardless of what’s sent over the protocol.  
It’s interesting that after switching to heat mode and setting 17°C, the unit then allows that temperature when switching back to cool — this seems like a quirk of the unit itself.

A few things you could check:
- Can you set 16°C in cool mode using the **original Samsung remote control**?  
  If yes, could you enable debug logs in ESPHome and share the logs captured when setting 16°C from the remote?  
  That would help us compare the packets and see if the remote is using a different mechanism or undocumented message.

Conclusion: ESPHome appears to work as expected, but this might be a firmware restriction on the unit itself.  
You could also check the indoor unit service manual for any temperature limit settings (sometimes there are DIP switches or service menu options to configure this behavior).

Thanks again and looking forward to your feedback!

BlairC1, 2025-07-17:
Hi Omer,

Yep I can set 16 using the original remote and could set it using the stock wifi module (removed now). I've done some digging through the service manual and it doesn't seem to have any defined service codes related to max/min set temperatures, all I can see is option codes for different modules. See attached log when using the remote to set 16 deg.

[logs_bedroomsamsungac_logs-2.txt](https://github.com/user-attachments/files/21282804/logs_bedroomsamsungac_logs-2.txt)

brazoayeye, 2025-07-17:
If you connect both esp and stock wifi modules you can see moving messages using log, so you can understand exaclty what stock module send to set the 16°C

BlairC1, 2025-07-18:
Here is the log of it when controlled by the Samsung wifi module

[logs_bedroomsamsungac_logs-3.txt](https://github.com/user-attachments/files/21307346/logs_bedroomsamsungac_logs-3.txt)